sarchlab/akita #41: Integrate Vis4Mesh

syifan, 2023-09-08:


yoyo115956, 2024-06-12:
Hello, I see that the two files meshnetworkinfo.go, meshnetworktracer.go store data in json format related to the data needed by Vis4Mesh project's Visdata, but there is no code found out there that calls the methods implemented here, may I ask how this is used?

ueqri, 2024-06-14:
Please refer to: (1) the comment here https://github.com/ueqri/vis4mesh/issues/8#issuecomment-2163664368 for where the NoC tracer is called/used, and (2) [Akita & MGPUSim Book: Chapter 2. The Akita Simulator Framework: Hooking and Tracing](https://syifan.notion.site/2-4-Hooking-and-Tracing-Ongoing-0b947051beb74868a2a74de81bafe40c?pvs=4) for how tracing works in the Akita framework. Thanks!