sarchlab/akita #89: NVLINK bandwidth performance in acceptance

MaxKev1n, 2024-03-13:
[image]

[image]

I found that the NVlink bandwidth performance in the acceptance is similar to the bandwidth performance of PCIe.

[image]

In addition, I change the number of link from 1 to 4 (the third parameter of *connector.ConnectDevicesWithNVLink*) but the performance doesn't change.

syifan, 2024-03-13:
@nichosta Do you have some time to help us looking into this problem? We can discuss it in our next meeting.

nichosta, 2024-03-15:
I will look into it when I have a chance.