sarchlab/akita #294: Record & Query Component Connectivity at Runtime

syifan, 2025-09-02:
## Motivation

Today Akita does not keep a canonical record of how components are connected once a model is built. That gap makes it harder to:

1. Configure components: e.g., an L1 cache doesn’t know who its next-level cache is without bespoke builder-side dependency injection. That tends to produce nonstandard per-project wiring code.
2. Observe the system: users (and tools) can’t easily answer “who is connected to whom?” at runtime. AkitaRTM and Daisen cannot draw a reliable connection map because there’s no single source of truth for edges in the topology.

Akita’s core already exposes `sim.Port`, `sim.Connection`, and implementations such as `sim.DirectConnection` and `noc/wiring.Wire`. Ports implement `SetConnection` and connections provide `PlugIn` / `Unplug`, but there’s no central registry to query peer ports on the same connection.

Recent changes also introduced a first‑class `Simulation` struct, which is an ideal place to keep lightweight indexes and expose public iterators/APIs.

## Proposal (minimal, backwards‑compatible)

lchangxii, 2025-09-03:
This can help users understand the entire system and how each unit connects with the others. In addition, it can assist users in identifying bugs related to component connections. 
In my view, it would be even better if we could automatically generate an interactive figure—allowing users to zoom in or out to explore connections between and within components—rather than relying on a JSON file.

syifan, 2025-09-03:
> This can help users understand the entire system and how each unit connects with the others. In addition, it can assist users in identifying bugs related to component connections. In my view, it [...]

Yes. The visualization (interactive figure) can be implemented in AkitaRTM or Daisen.

duoduoyeah, 2025-09-04:
My thought here is on observing the system,i.e. answering “who is connected to whom?” and visualizing the component-level topology within the simulation. This is distinct from examining individual component configurations.

## Limitations of Using Connections for Observation

I've previously attempted to use connections for this purpose but encountered issues.

**The problem:** for one specific port like  `comp_a.port`, we can not know which ports actually connect to it.

Consider the following scenario:

**Components:** `comp_a`, `comp_b`  
**Ports:** `comp_a.port`, `comp_b.port`  
**Actual connection:** `comp_a.port` and `comp_b.port` are directly connected.

For `comp_a.port`, there exists a connection object `conn_large`. However, `conn_large` may include many other ports that are not actually connected to `comp_a.port`, making it difficult to accurately observe the true topology.

So, If we still want to use conn for observing aspect, we need to redesign the conn part to solve this issue.

## My design on this "connection stuff"

### mgpusim side
Two things need to be collected:

1. **All components in the simulation**  
   These components are already been cached by `Simulation.components`.(There are some tiny bug which I will PR later)

2. **Port connections from Messages**  
   For every port of each component, record its incoming and outgoing ports.  
   This information can be deduced from the `Message`.  
   The process adds an overall **O(N)** time complexity (where *N* is the number of messages) on top of the base simulation. Also, not every simulation need to do this.

### akita side
Create two new tables in the sqlite:
these two tables contain the necessary data to create the topology map for our gpu.

```go
type TopologyPortEntry struct {
	Port      string `json:"port" akita_data:"unique"`
	Component string `json:"component" akita_data:"index"`
}

type PortConnectionEntry struct {
	SourcePort      string `json:"from_port" akita_data:"index"` // Source port ID
	DestinationPort string `json:"to_port" akita_data:"index"`   // Destination port ID
}
```

## Another proposal
Currently, all traces are stored in a single SQLite file. However, there is no corresponding frontend tool (such as Daisen) that can visualize the data directly from this file.  

For better usability, we could provide a simple web interface. For example, the user could run a command like `npm start`, and then view all the trace information in a unified web application.

This unified web application could include:
1. Basic simulation metrics
2. Daisen
3. Topology map allowing users to zoom in or out to explore connections between and within components.

syifan, 2026-10-14:
Covered by what v5 now has:

- **Observation.** The topology recorder writes every registered port with its owning component and the connection it is plugged into (the `port` table, `sim/topology_recorder.go`). Ports that share a connection are that connection's endpoints, so tools can draw the connection graph from it.
- **Configuration.** The command-processor case is solved without a runtime lookup. Per #492, peer addresses go in the Spec as `messaging.RemotePort` or `[]messaging.RemotePort` fields.

Read-only `Components()` and `Ports()` on `*sim.Simulation` are part of #538. Closing.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-10-14 by syifan